sdr-enthusiasts/docker-planefence #18: Plane-alert tweets everything every time

kx1t, 2021-02-25:


Sportsbadger, 2021-02-25:
I was thinking about this - maybe you could either rate limit, only tweet once per X. Or you could have a check to say... "if tweeted about plane Y in period X, then don't tweet"

kx1t, 2021-02-25:
It's a bug. The desired/implemented behavior is that plane-alert (not planefence!) ignores multiple occurrences of the same hex id during the same day. This works, but for some reason i received a flurry of Twitter DMs.

kx1t, 2021-02-26:
reviewed, made minor changes, and waiting for reproduction of the bug

kx1t, 2021-03-05:
made more changes that should fix it. 
Most of the issues were related to `read -r line < file.csv` .. `read -ra record <<< "$line"`  if $line contains unescaped special characters. Weird... the `-r` option should have read in the text "raw" - maybe it get mangles or expanded somewhere downstream

closed 2021-03-05 by kx1t