sdr-enthusiasts/docker-planefence #19: List of aircraft to ignore

Sportsbadger, 2021-02-25:
On sunny days the local flying club are setting off the fence repeatedly as they fly in and out all day. Would it be possible to define a list of 'ignore' aircraft that Planefence would NOT tweet about. Or even ignore altogether and not be logged.

kx1t, 2021-02-25:
An ignore list would be relatively easy to implement. Another option could be to add a ignore_time parameter that only considers an aircraft once every x seconds.

Do you have any preference? Either / or / both?

In either case, if prefer to implement this "globally" (i.e., filter the aircraft right when we are processing the input) rather than only for tweeting.

Sportsbadger, 2021-02-25:
I like both ideas - the ignore list and the ignore_time parameter. I think in conjunction with each other I think they could cut down on tweet repetition. And I understand and agree about implementing the features globally.

kx1t, 2021-03-05:
Implemented - ready for pull

closed 2021-03-05 by kx1t