sdr-enthusiasts/docker-planefence #24: Only trigger planefence when specified aircraft are seen - Accept list

Sportsbadger, 2021-03-07:
Think of it as the opposite of the ignore list. In this case planefence would not trigger/log/tweet unless the aircraft seen was on the accept list. In my use case I want to trigger/log/tweet only on helicopters over my home city. I appreciate this is similar to how plane alert works, but as I want these tweets to be public I think it is a different use case.

kx1t, 2021-03-18:
Thinking about this some more...
I rather add an option for Plane-Alert to tweet _either_ as DM _or_ as normal tweet. That way, we don't muddy the purpose of the two sub-applications: Planefence = limited area, all aircraft; Plane-Alert = feeder-wide area, specific set of aircraft.

Would that be OK?

(hmmm I guess I already violated that separation by adding "PF_PARANGE" to enable a max range for Plane-Alert, but alas...)

Sportsbadger, 2021-03-19:
That is a far better implementation of the idea - I love it.

kx1t, 2021-06-11:
Tweeting vs DMing Plane-Alert messages was implemented and deployed

closed 2021-06-11 by kx1t