sdr-enthusiasts/docker-planefence #27: Add tags to plane alert list

Sportsbadger, 2021-03-07:
Could a user defined tag be added to the plane alert webpage - I have defined tags (choppa, specialist, fighter, absolute unit) against my watch list and it would be nice to see them on the alert page.

kx1t, 2021-03-10:
Where did you define these tags? 
Probably the easiest (and best backwards compatible) way would be to add another field to the `plane-alert-db.txt` file
... right now it looks like this:

`A2B2D2,N273NE,NE Life Flight,EuroCopter 45`

What about if we allow this:
`A2B2D2,N273NE,#NE Life Flight,#EuroCopter 45,#My Custom Tag`

In this case, I could display a new column on the website with the tags `#NELifeFlight #Eurocopter45 #MyCustomTag` and also add this to the Tweet DM.

Note that the max length for a DM is 10,000 characters, so we probably won't run into length issues (like we did for regular tweets).

@Sportsbadger  please let me know your thoughts on the implementation proposal.

kx1t, 2021-03-17:
Implemented and deployed to `dev`
https://github.com/kx1t/docker-planefence/commit/8d21d076f90e5eeabcbc45cb5673bd14df8af837

closed 2021-03-17 by kx1t