sdr-enthusiasts/docker-planefence #38: Allow more detail/columns in alert list

Sportsbadger, 2021-03-17:
Now that I'm putting together alert lists I have realised that it would be useful to have more data/columns in the alert list, even if they are not used by PA. For example I store the data in this format, 

ICAO,Ident,Operator,Type,Mil/Civ/Pol,Tag,Category

As the list grows it would be useful to have this extra data the user knows why the plane is on the list, and to allow for better management of what aircraft should be altered on. 

In short - richer tagging of the alert list.

kx1t, 2021-03-17:
Currently, you are already able to add on additional fields beyond the Plane Type.

I suggest we do the following:
- the first line of the `plane-alert-db.txt` file is the index to the table
- the Plane-Alert table will always show ICAO/Transponder ID, Tail, Owner, Equipment/Planetype
- For any additional fields, if the field name starts with a "#" it will NOT be showed, or if the field name does not start with "#", it will be shown in the table
- I probably have to limit the number of extra fields to 4 or 5 or so (for a total of 8-9 fields shown) - to ensure it fits on the screen (and to avoid things go wild).

Let me know if this is what you want?

Sportsbadger, 2021-03-17:
That sounds absolutely perfect. Thank you.

kx1t, 2021-03-17:
Implementation pushed to dev:
are you ready to test?
- update the header of plane-alert-db.txt:
```
#Transponder ID,Tail,Owner,Equipment,Custom 1,#Custom 2,Custom 3,#Custom 4,Custom 5
```
- Now, it will print the NAME of the headers into the table
- Always start the line with a "#"
- It ALWAYS use fields 1-4 even if the fieldname starts with "#" (which is will strip off)
- if you add any of the custom fields -- they WILL show up in the table as long as the fieldname does NOT start with "#". In my example -- it will show fields "Custom 1", "Custom 3", and "Custom 5"
- You can add up to 7 custom fields (for a total of 11 fields). That's a random limitation - if that's not good enough, let me know
- You can also do LESS than 7 custom fields or no custom fields -- if you don't add them to the header, they will be ignored
- Then... edit the rest of the entries in this file and add info into the custom fields
For example:
```
A0B674,N145NE,NE Life Flight,EuroCopter 45,hello 1, hello 2,hello 3,hello 4,hello 5
A2B2D2,N273NE,NE Life Flight,EuroCopter 45,hello a, hello b,hello c,hello d,hello e
```
results in this: 
https://ramonk.net/planefence-dev/plane-alert/

kx1t, 2021-03-17:
Implemented and deployed to `dev`
https://github.com/kx1t/docker-planefence/commit/8d21d076f90e5eeabcbc45cb5673bd14df8af837

closed 2021-03-17 by kx1t