sdr-enthusiasts/docker-planefence #53: OpenAIP layer on Planefence Map

rikgale, 2021-03-27:
Hey kx1t, 

Suggestion - would it be possible at to add the OpenAIP layer to the planefence map?

Many thanks
rikgale

kx1t, 2021-03-27:
Should be possible, I'll look into it. I haven't played around with the OSM
layers much in the last few months.

kx1t, 2021-04-02:
This is implemented in https://github.com/kx1t/docker-planefence/pull/58/commits/61c56f2f2510ba159876b749b004d8737894e003 and released to `main` via PR https://github.com/kx1t/docker-planefence/pull/58.

Configuration -- add `PF_OPENAIP_LAYER=ON` to your `~/.planefence/planefence.config` file and the heatmap on the website will show the OpenAIP overlay. If `PF_OPENAIP_LAYER` is absent or set to anything but `ON`, the OpenAIP layer won't show.

This is also documented in the example `planefence.config` included with the docker distro.

Note that loading of this layer can be slow, especially the first time when the data is not yet in your browser cache. This is caused by OpenAIP.net not rendering the overlay tiles as fast as we may wish, and unfortunately it's not something that is controlled or influenced by PlaneFence.

Note also that this was tested with a direct, non-secure connection to Planefence as well as with a secure connection via a reverse web proxy. Rendering and latency was similar in both cases.

closed 2021-04-02 by kx1t