sdr-enthusiasts/docker-planefence #74: Links to flight plans

hakan42, 2021-05-12:
Could we link the "Flight" column of the Flights in Range table to a service that provides flight plans? FR24 maybe?

I am curious where all those bombardier jets come from which fly right over my house on their way to EDMO

*  https://www.flightradar24.com/airport/obf

kx1t, 2021-05-14:
Looking into this. The actual programming part is relatively simple.
The following needs to be resolved first before I can implement this:

- I'd need a service that can reliably look up flight plans by (either) ICAO Hex ID, or Flight number (or tail if there's no flight number).
- Then, we also need to make sure that a link to that service doesn't violate the T&C's of the website we're linking to.

kx1t, 2021-05-14:
I could use a similar link as is used by PiAware ... for example: https://flightaware.com/live/modes/A44C70/ident/DAL2267/redirect

kx1t, 2021-05-14:
Initial implementation now available in `docker: kx1t/planefence:dev`
Commit f8332a30987d0a2e0d19ea993482cd97c2bdc5c9

@hakan42 - please download the `kx1t/planefence:dev` container and see if you like it. If there are any issues, please feel free to reopen this Issue.

Thanks! Ramon

closed 2021-05-14 by kx1t

hakan42, 2021-05-15:
Thank you, this is excellent. Could you perhaps create an amd64 image too or should I wait for this change to find its way to the main branch?

kx1t, 2021-05-15:
Actually, I promoted / PR'd it into the main branch yesterday afternoon, so
it should be in the `latest` tag.

Let me know if you see anything strange running the multi-arch build on
amd64 as I have no means to test that here.