sdr-enthusiasts/docker-planefence #75: Send events over MQTT too

hakan42, 2021-05-14:
Send the event to be tweeted about as a json-formatted message to a MQTT server, a client of which in turn can decide whether to tweet about it or notify the home automation or to water plants :)

- https://github.com/jpmens/mqttwarn

This could be in addition to the tweet setup, or the tweeting part could be extracted into its own container.

If a running total of the current day / week / whatever could be included in the message, this could tie in well with #21

kx1t, 2021-05-14:
How would that work? Where would the `mqttwarn` server be running? On the same Rpi? In the Planefence container?
Generating a JSON formatted message isn't too hard, but I need some help / research on how to send that message to a MQTT server.

hakan42, 2021-05-14:
The mqttwarn server would run... wherever you want it to. I would assume that someone who would run such a setup would already have a server where mqttwarn is already available.

I would avoid running the tweeter in the same container as planefence, I tend to keep each container as small as possible. For people who want to have the same functionality as now without changing too much, we could create a compose file which starts planefence, mosquitto and mqttwarn.

I could set up a documentation how to create a tweeting agent, there are examples in the mqttwarn docu itself.

kx1t, 2021-05-14:
Right now, tweeting is initiated and timed during the Planefence run, as
there is some timing involved to ensure that enough information is
available to tweet. In reality, it's extremely lightweight and uses Twurl
(basically a Ruby based Curl wrapper) to send the tweets. I have no plans
to discontinue or separate this tweeting service, as I think most people
won't want to set up MQTT service just to send a tweet.

Alternatively, I have also been looking into adding Telegram and other
notification services, however, for those, maybe using a MQTT service with
Telegram notification plugin would be better.

As for MQTT, ideally, within Planefence, I would create the JSON for a
notification (very easy) and then use CURL or something similar to push the
notification. As such, if you can point me at the documentation
that describes how to push JSON based notification to MQTTWarn, that would
save me a bunch of research time.

Note that the implementation for this will take a bit of time -- there are
some high priority projects I need to finish first before I can start
looking into setting up a MQTT test environment.

hakan42, 2021-05-15:
You could send your message to mosquitto using the command line client. Steve has many good examples on his guide.

For the docker image, the command line tools should be included in the "mosquitto-client" package.

- mosquitto_pub -h <hostname-or-ip> -m "{ lat: '47.11', lon: '08.15', test: 'foobar' }" -u <username> -p <password>
- username and password are optional in most of the installations
- http://www.steves-internet-guide.com/mosquitto_pub-sub-clients/

I guess you would want to include the path to the screen shot in the message as well, and whoever wants to consume the message would need to find their own way to get the image file to the mqttwarn container.

hakan42, 2021-05-16:
Actually, @mikenye created some shell scripts to push json-formatted data to an MQTT server too:

https://github.com/mikenye/docker-adsb-to-mqtt/tree/master/rootfs/usr/local/bin