sdr-enthusiasts/docker-planefence #79: User customisable links between PF page and PA page

rikgale, 2021-05-26:
Hi Kx1t

Would it be possible to have an HTML link between the PF and PA pages (and visa versa)? With the links defined in the .config file. 

Rational for the links - it would make for smoother navigating for those users that have both installed. Rather than going into PF and wanting to go to the PA page the user has to back out of the PF page and then navigate to the PA page. If there was a link between the two it would make for a nice user experience.  If users don't activate PA then hide the link

Rational for user defined links - as some users run a website behind your reverse proxy container, other webserver or just run locally each address would be different so having the user enter there own link would allow for flexibility.

Thanks
R

kx1t, 2021-05-26:
SHould be easy -- I will add it to the website bullet list if `PF_WEB_PA_LINK` contains a URL.
By default, the URL can be "plane-alert" which will automatically point at the sub-page with Plane Alert -- no need to add an absolute URL.

kx1t, 2021-05-26:
Implemented in `Dev` as of commit ae0f33e0df06b784695f486e45664231b39f5900. Add this to your `planefence.config` file:

```
# ---------------------------------------------------------------------
# PA_LINK is a URL that points from the PlaneFence web page to the Plane Alert webpage.
# Similarly, PA_PF_LINK is a URL that points from the Plane Alert page to the PlaneFence web page.
# The default values will work in most circumstances
# If empty, it's omitted
PF_PA_LINK="plane-alert"
PA_PF_LINK=".."
```

Feel free to pull the `kx1t/planefence:dev` container and test it

closed 2021-05-26 by kx1t