sdr-enthusiasts/docker-planefence #81: (additionally) tweet height above ground level

hakan42, 2021-06-04:
Currently, planefence tweets whatever height has been reported by readsb (which usually is pressure altitude: https://www.aopa.org/training-and-safety/active-pilots/safety-and-technique/weather/density-altitude#:~:text=Density%20altitude%20is%20pressure%20altitude,airplane%20%22feels%22%20its%20flying. )

If we could have a configurable delta to the height fetched from readsb, we could **additionally** tweet the height above ground for people who are not well versed in aerospace units. The main purpose is for my neighbours to be informed of any flights below 1000 feet AGL, in units and altitude they are used to (mainly, meters above ground)

If possible, and there is still enough space, this should be in addition to the text alreay in the tweet, with a very short remark that this is AGL.

kx1t, 2021-06-04:
For the tweet, I suggest that we either report MSL or AGL altitude, but not both. The reason is that we are continuously struggling to keep the tweets within the 280 character limit set by Twitter.

My idea was to make the barometric to AGL correction at the point of the initial parsing of the readsb (SBS) data, so when users set the PF_MAXALT parameter, this will be in MSL or AGL, if that makes sense?

hakan42, 2021-06-04:
Would you need an additional data source for the pressure at the location and moment of the tweet (which might require contacting the nearest airport if they have a machine readable data source available) or just go with the standard pressure of 1013hPa? For my purposes, the standard would be good enough as I don't plan to tweet about any passenger airlines flying at FL390 :)

Apart from that, your approach is perfect for me.

kx1t, 2021-06-04:
No, I'd simply add a config parameter for the user to set what the local
elevation is, and subtract that from the reported barometric altitude.
I don't want to make it more complex than it has to be.

kx1t, 2021-06-04:
I've implemented the following, subject to testing (which hasn't been done yet):

- added an optional parameter "PF_ELEVATION" where the user can add the local ground elevation. 
- unit is whatever the user defined altitude unit is for Planefence
- if `PF_ELEVATION` >0 (I could rethink this for people in Holland :) ), the displayed altitude will be `baro_alt - elevation`
- if `PF_ELEVATION` >0, the tweet will append "AGL" behind the reported altitude, and if not, it will append "MSL"
- The PlaneFence website has an additional note to explain if the Minimum Altitude column is MSL or AGL.

Note -- any changes by the user to the `PF_ELEVATION` parameter will be prospective only. This means that Planefence will NOT go back and recalculate previously reported altitudes. This will make intra-day changes inconsistent on the web page and in the CSV file, where the reported minimum altitude depends on the `PF_ELEVATION` setting at the moment the record was created. Although this is a minor inconvenience, there are no current plans to address this as users generally won't change this parameter very often.

kx1t, 2021-06-04:
@hakan42 the current `dev` build contains a candidate solution. Simply add `PF_ELEVATION= xxxx` to your planefence.config file, where xxxx is the local elevation in whatever unit you are using with planefence.

There are prebuilt `kx1t/planefence:dev` images in docker for `armhf` and `arm64`; if you need the `amd64` image, you'd have to build it yourself using the Dockerfile in the `dev` branch of this repo.

Let me know if there are any issues.

closed 2021-06-11 by kx1t