sdr-enthusiasts/docker-planefence #101: **Feature Request: Plane Alert Webpage -  addition of Silhouette of aircraft**

rikgale, 2021-06-19:
Insert a new column next to the ICAO (tag3) column and use the information from this ICAO column to pass to a variable and use this to create a link to a file (files probably best stored within the container). This would then display a small image of a silhouette for each type of a/c on each line.

[image]

Please excuse the crude hack job.

All the silhouettes I use and make are of the name format ICAOTypeCode.bmp and are 85*20 px in size.

eg Boeing C-17 Globemaster = C17.bmp

Ideally has to be a bmp file as this is the standard used by VRS for which I have started making silhouettes.

kx1t, 2021-06-24:
Do you have a sample of these files?
Also, are they hosted somewhere online (ideally as a ZIP / TAR / TGZ or otherwise compressed file), so I can tell the container to "pull" them on a regular basis?

If anything, I would want to implement something like this (which is similar to the way the plane-alert-db.csv updates work):

- Upon start and every XXX hours afterwards, and when it detects a change to the planefence.config file, the container will kick off a "remote update" attempt
- It would download and install (overwrite existing images) a single file with all the images
- BMP would be OK for now -- it's "somewhat" supported across current browsers.
- the FILE NAME would have to correspond to the ICAO Plane Type, for example "B744.bmp". We'd need to settle on filename capitalization - I don't care what it is, as long as it is consistent. I think that is what you are already suggesting in your request.
- Empty files for ICAO Plane Types that aren't yet supported isn't needed - I'd simply put the picture in a field cell if it is available, and leave the field cell empty if it's not available.

Also, keeping in spirit with the existing lay-out, I may want to put the image in the FIRST COLUMN, but I am open for suggestions and can be convinced otherwise. (An image in the middle of a table isn't too visually appealing, in my opinion.)

rikgale, 2021-06-24:
"Do you have a sample of these files?" Yes - I have them in my github repo https://github.com/rikgale/VRSOperatorFlags/blob/main/Silhouettes.zip

ZIP needs an update as have created a fair few more of late, but location will be the same.

"the FILE NAME would have to correspond to the ICAO Plane Type" - They are all aligned with ICAO type codes so they work with VRS and Basestation (This is a must for other users including myself). All I create will have the file name aligned accordingly. The only exceptions to this would be where an a/c doesn't have an ICAO code, but if you are pulling the ICAO type from sportbadgers list then I can make sure they all match as I've been populating the ICAO type codes for him.

Keep them to the left if that works for you :)

rikgale, 2021-06-24:
FYI SB and I have a way of highlighting missing silhouettes in the backend of his list so we will be focusing on getting those completed first.

kx1t, 2021-07-01:
Implemented in https://github.com/kx1t/docker-planefence/commit/281c365940222681a1414ffbbcb24c003bcaebd6

closed 2021-07-01 by kx1t