sdr-enthusiasts/docker-planefence #112: Every airline / owner starting with N appears to link to FAA tailno. lookup page

sgofferj, 2021-09-13:
Example:
https://planefence.gofferje.net/planefence-210913.html
`4ACA4A    NAX2524    Norwegian Air Shuttle`

sgofferj, 2021-09-14:
Addition:
Planefence seems to also use the callsign to identify the operator which doesn't always work.
https://planefence.gofferje.net/planefence-210913.html
Line 27 `46788F N21 U S DEPT OF COMMERCE`
46788f is a CASA C-295 medium transport plane operated by the Finnish Air Force. It was running "November 2-1" as a callsign. Planefence identified it as the deregistered/destroyed plane with the tail number N21 of the US DoC...

Suggestion:
Implement the same database that tar1090 uses.

kx1t, 2021-09-14:
I will add a check that the Hex ID needs to start with an "A" as an extra qualifier for the FAA lookup. In the end, I am using the Mictronic database, which what tar1090 uses, but it's only used as a fallback at this time.

kx1t, 2021-09-14:
This is what the (updated) check looks like. Only if this expression is true, it will show the FAA link:
```
[[ $CALLSIGN =~ ^N[0-9][0-9a-zA-Z]+$ ]] && [[ "${CALLSIGN:0:4}" != "NATO" ]] && [[ "${NEWVALUES[0]:0:1}" == "A" ]]
```
In here, `CALLSIGN` contains the callsign, and `${NEWVALUES[0]}` contains the hex ID

closed 2022-08-03 by kx1t