sdr-enthusiasts/docker-planefence #140: [Enhancement] Send alert to push notification aggregator service

SingingDwarf, 2022-02-09:
Suggestion for enhancement - send notifications to a self hosted push aggregator service, such as Apprise (https://github.com/caronc/apprise) which would allow users to send notifications to services of their choice, but with only a single integration point for this project to maintain.

SingingDwarf, 2022-06-10:
@kx1t Sorry for the radio silence on this, unfortunately 'life' has been in the way recently.

Is the work you refer to above using the NOTIFICATION_SERVER setting in the config file? I cannot find any documentation relating to this setting, so assume that it is new/undocumented feature currently?

beercity, 2024-09-11:
adding Ntfy (ntfy.sh) as an option would be great as well.

kx1t, 2025-02-13:
We’ve all but abandoned implementing using a notification service like Apprise. At the moment, Planefence and Plane-Alert support X (barely, you need to pay for API access), Mastodon, Discord, BlueSky, and MQTT notifications. I’m sure that there are MQTT-to-Telegram handlers if you want to go that route. 

Closing it as “not planned”

closed 2025-02-13 by kx1t

driftfox, 2026-08-27:
Hi folks,

There is a related proposal open now for generic HTTP webhook POSTs: #382, with a PR at #381.

If that lands, you should be able to cover a lot of what was asked here by pointing Planefence / Plane-Alert at an HTTP endpoint you control:

- ntfy works well already (text or JSON), which was something @beercity mentioned
- Telegram should be doable the same way @kimfr described, with a curl-style POST to the Bot API and a template for the message body
- If you still want Apprise in the middle, you can often POST into an Apprise API (or similar) and let that fan out to other services

I have been testing the PR locally against ntfy. Happy to take feedback on the issue or the PR.