sdr-enthusiasts/docker-planefence #158: Tweet map screenshot has black background

chevyman142000, 2022-08-16:
All posted tweets from my setup are being posted with black map backgrounds. You can see what they look like here: https://twitter.com/msnairspots Planefence and the screenshot container are running on a Synology nas. Let me know what other details I can get you to help troubleshoot this. Thank you!

Kevin10187, 2023-02-14:
This is currently happening on my setup. What was the fix?

chevyman142000, 2023-02-22:
> This is currently happening on my setup. What was the fix?

Sorry for the delayed response. I had to mess with the zoom settings on the screenshot container. I ended up setting it to this and it started working: zoom=11&hideSidebar&hideButtons