sdr-enthusiasts/docker-planefence #159: Discord Alerts Throwing Error

cbattlegear, 2023-01-13:
Discord alerts are currently failing with an error:

`requests.exceptions.InvalidSchema: No connection adapters were found for "['WEBHOOK_URL_HERE']"`

This seems to be an issue wiht the discord-webhooks library all up though as running discord_webhook in the container via:

`docker exec -it planefence discord_webhook -u WEBHOOK_URL_HERE -c Test`

is also throwing an identical error. 

Currently can be resolved by iterating through the list of webhook URLs and converting to tuples, then iterating through the tuples to send the alerts. I have code I will work on creating a pull request for.

kx1t, 2023-01-17:
Hi - you are absolutely right that this is caused by a change to the Discord library.

I merged in your code (and reverted the inclusion of the old Discord library to the latest in the Dockerfile) but it doesn't seem to work. Could you take a look?

Also - could you pls submit any PRs for this to the `dev` branch -- this branch has the latest code to notify to Mastodon, and as soon as the Discord notification issue is fixed, I'll PR the entire `dev` branch into `main`.

Thanks again for you help here!

cbattlegear, 2023-01-17:
Totally! Sorry I am terrible at reading docs, just went hard on fixing it locally. I will get some testing done on my side and PR it with the new discord library

kx1t, 2023-01-17:
Note that I recently added some additional files to the `dev` branch with Discord in the name -- you can totally ignore anything related to Discord that is not in Python.

cbattlegear, 2023-01-19:
Should have a fully working build in pull request #162 that resolves these issues

kx1t, 2023-01-28:
Thanks again for your help here. I had 2 fixes for this issue that I consolidated and included as part of the larger PR / release of yesterday. Should be resolved now.

closed 2023-01-28 by kx1t