sdr-enthusiasts/docker-planefence #208: ADSB Report Generation

Mike-D-424, 2024-10-26:
Please remove or relocate if this is posted in a different place.  I tried to email you; however, it bounced back.

I wish to find a magic way for either my existing ADS-B feed or from the internet to create and (preferably e-mail me) a weekly report, including aircraft ID, speed, heading, date, time, altitude, etc., of any aircraft lower than 1000 ft, within two miles of my home.   

I want this because I am tired of helicopters making my house shake since I live under their approach to a local airport.

I am not a coding or pi expert; therefore, a more straightforward and easier-to-implement solution is preferred.

kx1t, 2024-12-09:
Hi - I will add this as a potential future feature request, but we're currently not considering to implement this. The data that you would need for this is already available on your Pi, in whatever directory you mapped for "html":
- `....html/planefence-yyyymmdd.csv` has the Planefence data (in CSV format, you can import this in Excel/Google Sheets/MacOS Numbers) for that date
- `....html/plane-alert.csv` has the Plane-Alert data (from the beginning of times), you can similarly import this into any of the spreadsheet programs

We also added MQTT notifications; if you have a MQTT broker, you can integrate this with your Google Smart Speaker, etc.
All of this is _not exactly_ what you asked for, but maybe it helps.

Mike-D-424, 2024-12-09:
Thank you for answering. 

Could you please give me some idea of how to find the directory where the files are?

I am NOT a tech guy (especially with the pi)

Thank you again!

Michael

kx1t, 2025-02-13:
In your `docker-compose.yml` file, you defined 2 directory mappings in the `volumes:` section under `planefence`. Ones of them maps to a html directory. The left (of the “:”) hand part is the directory in which the `planefence-yymmdd.csv` files are stored. 
That same directory has a subdirectory called `plane-alert`. In that subdirectory, you’ll find `plane-alert.csv`. 

Not that we are in the process of adding export buttons to the table; you can preview this at http://kx1t.com/planefence-dev/ - this may help you as well

closed 2025-02-13 by kx1t

Mike-D-424, 2025-02-15:
Ramon,

I realize you are a fellow ham and an unpaid volunteer. I'd appreciate it if you would consider allowing me to screen-share my Pi with you so you can show me what you are referring to.

I'll be around most of the weekend.

Please let me know what works best for your schedule.

'73 N2OPG

Michael Del Greco
56 Woodland Drive
Woodland Park, NJ 07424