sdr-enthusiasts/docker-planefence #210: 403 Forbidden when clicking monitored aircraft Alert List

kimfr, 2024-11-13:
In a newly deployed planefence container - when I click on the monitored aircraft Alert List - I get a 403 Forbidden.

Looking at the link via the page http://<ip>:8088/plane-alert/ - I can see it points to http://<ip>:8088/plane-alert/alertlist.txt

Looking inside the planefence container, I get:
```
# find / -name alertlist.txt
/usr/share/planefence/html/plane-alert/alertlist.txt
/usr/share/plane-alert/html/alertlist.txt

# ls -la /usr/share/planefence/html/plane-alert/alertlist.txt
lrwxrwxrwx 1 root root 58 Nov 13 12:34 /usr/share/planefence/html/plane-alert/alertlist.txt -> /usr/share/planefence/persist/.internal/plane-alert-db.txt

# ls -la /usr/share/planefence/persist/.internal/plane-alert-db.txt
-rw-r--r-- 1 root root 3767636 Nov 13 12:34 /usr/share/planefence/persist/.internal/plane-alert-db.txt

# ls -la /usr/share/plane-alert/html/alertlist.txt
-rw-r--r-- 1 root root 10545 Aug 16 22:17 /usr/share/plane-alert/html/alertlist.txt

```
Both files can be read from within the planefence container:
```
tail -n1 /usr/share/planefence/html/plane-alert/alertlist.txt
E940FB,FAB-002,Gov of Bolivia,Dassault Falcon 50,FA50,Gov,Government,Must Be Nice,Free And Fair Elections,Dictator Alert,https://w.wiki/B3LJ,https://cdn.jetphotos.com/full/6/55153_1575059321.jpg,https://cdn.jetphotos.com/full/5/83758_1487098979.jpg,https://cdn.jetphotos.com/full/5/17432_1487090281.jpg,

tail -n1 /usr/share/plane-alert/html/alertlist.txt           
A2B2D2,N273NE,NE Life Flight,EuroCopter 45
```
I'm also a little confused as to why there are 2 different files?

kimfr, 2024-11-13:
It's NOT the plane-alert-db.txt that's the issue, it's working fine  - it's when built in "alertlist.txt" - ie. the list of used for Plane-Alert.

kimfr, 2024-11-14:
Left it running overnight and this morning it has now picked up 2 aircrafts so the list is being read and used - but when I click on the "list of monitored aircraft" - I still get a 403 Forbidden.
[image]
[image]

kx1t, 2024-12-09:
I've checked this, and I can't reproduce it. Closing this issue for now; if it appears again, feel free to reopen

jj358mhz, 2024-12-12:
I can reproduce it
[image]

kx1t, 2024-12-17:
I’ll look into it some more

closed 2024-12-17 by kx1t