sdr-enthusiasts/docker-planefence #211: Feature Request: RSS feed

random-robbie, 2024-12-07:
Hey,

I would like to get a RSS feed generator added to plane-fence.

I have a LED matrix that can only use RSS feed's i'd like the information to display on.

I've done the code for you to review.

```
#!/command/with-contenv bash
#
# planefence-rss.sh
# A script to generate RSS feeds from PlaneFence CSV files
#
# Usage: ./planefence-rss.sh 
#
# Copyright 2024 - Licensed under GPL V3

# Set paths - use the same as planefence.sh
PLANEFENCEDIR=/usr/share/planefence
[[ -f "$PLANEFENCEDIR/planefence.conf" ]] && source "$PLANEFENCEDIR/planefence.conf"

# Get today's date in yymmdd format
FENCEDATE=$(date --date="today" '+%y%m%d')

# Site configuration - you can modify these
SITE_TITLE="PlaneFence Aircraft Detections"
SITE_DESC="Recent aircraft detected within range of our ADS-B receiver"
SITE_LINK="http://your-planefence-url/"  # Replace with your actual URL
SITE_IMAGE="http://your-planefence-url/favicon.ico"  # Optional site image

# Function to encode special characters for XML
xml_encode() {
    echo "$1" | sed 's/&/\&amp;/g; s/</\&lt;/g; s/>/\&gt;/g; s/"/\&quot;/g; s/'"'"'/\&apos;/g'
}

# Function to generate RSS feed for a specific CSV file
generate_rss() {
    local csv_file="$1"
    local rss_file="${csv_file%.csv}.rss"
    local date_str="${csv_file##*-}"
    date_str="${date_str%.csv}"
    
    echo "Generating RSS feed for $csv_file"
    
    # Create RSS header
    cat > "$rss_file" <<EOF
<?xml version="1.0" encoding="UTF-8" ?>
<rss version="2.0" xmlns:atom="http://www.w3.org/2005/Atom">
<channel>
    <title>$(xml_encode "$SITE_TITLE - $date_str")</title>
    <description>$(xml_encode "$SITE_DESC")</description>
    <link>$(xml_encode "$SITE_LINK")</link>
    <lastBuildDate>$(date -R)</lastBuildDate>
    <image>
        <url>$(xml_encode "$SITE_IMAGE")</url>
        <title>$(xml_encode "$SITE_TITLE")</title>
        <link>$(xml_encode "$SITE_LINK")</link>
    </image>
    <atom:link href="$(xml_encode "${SITE_LINK}${rss_file##*/}")" rel="self" type="application/rss+xml" />
EOF

    # Process the CSV file in reverse order (newest first)
    if [[ -f "$csv_file" ]]; then
        tac "$csv_file" | while IFS=, read -r HEXCODE FLIGHT FIRSTSEEN LASTSEEN ALT DIST URL TWEET NOISE REST; do
            # Skip empty lines and comments
            [[ -z "$HEXCODE" || "${HEXCODE:0:1}" == "#" ]] && continue
            
            # Clean up flight number (remove @ symbol if present)
            FLIGHT="${FLIGHT#@}"
            
            # Create title and description
            TITLE="Aircraft ${FLIGHT:-$HEXCODE} detected"
            DESC="Aircraft ${FLIGHT:-$HEXCODE} was detected within ${DIST}${DISTUNIT} of the receiver"
            DESC="${DESC} at altitude ${ALT}${ALTUNIT} from ${FIRSTSEEN} to ${LASTSEEN}"
            
            # Add noise data if available
            if [[ -n "$NOISE" ]]; then
                DESC="${DESC}, with peak noise level of ${NOISE} dBFS"
            fi
            
            # Create item link - use the tracking URL if available
            ITEM_LINK="${URL:-$SITE_LINK}"
            
            # Calculate pub date from LASTSEEN (assumed to be in format "yyyy-mm-dd hh:mm:ss")
            PUBDATE=$(date -R -d "$LASTSEEN")
            
            # Write RSS item
            cat >> "$rss_file" <<EOF
    <item>
        <title>$(xml_encode "$TITLE")</title>
        <description>$(xml_encode "$DESC")</description>
        <link>$(xml_encode "$ITEM_LINK")</link>
        <guid isPermaLink="false">$HEXCODE-$FIRSTSEEN</guid>
        <pubDate>$PUBDATE</pubDate>
    </item>
EOF
        done
    fi

    # Close the RSS feed
    cat >> "$rss_file" <<EOF
</channel>
</rss>
EOF

    # Set proper permissions
    chmod 644 "$rss_file"
    echo "RSS feed generated at $rss_file"
}

# Find all CSV files and generate RSS feeds
for csv_file in "$OUTFILEDIR"/planefence-*.csv; do
    [[ -f "$csv_file" ]] || continue
    generate_rss "$csv_file"
done

# Create/update symlink for today's feed
today_csv="$OUTFILEDIR/planefence-$FENCEDATE.csv"
today_rss="$OUTFILEDIR/planefence-$FENCEDATE.rss"
if [[ -f "$today_csv" ]]; then
    # Make sure today's RSS exists
    [[ ! -f "$today_rss" ]] && generate_rss "$today_csv"
    
    # Create/update the symlink
    ln -sf "planefence-$FENCEDATE.rss" "$OUTFILEDIR/planefence.rss"
    echo "Updated symlink planefence.rss to point to today's feed"
fi
```

The script will:

- Generate an RSS feed for each CSV file found
- Include the date in the RSS feed title
- Create/update a symlink for today's feed
-  Maintain separate RSS feeds for historical data

kx1t, 2024-12-17:
Great idea! Can you implement this in a branch and test it?
I would do the following:

- save the script in /usr/share/planefence (make sure it’s executable when you commit)
- Call the script from /usr/share/planefence-planefence.sh somewhere close to where that script calls planefence_notify.sh
- Enhance the cleanup script so it deletes old rss files in line with the expiration policy (like this/here): https://github.com/sdr-enthusiasts/docker-planefence/blob/e79e5aa6223c2d046bb31977efb5ed1fc54a0d4a/rootfs/etc/s6-overlay/scripts/cleanup#L56)

That way your rss will be updated every time Planefence runs

Then you should consider adding something similar to the plane-alert script as well :)

kx1t, 2024-12-20:
hey @random-robbie -- I implemented this for Planefence only following your script in https://github.com/sdr-enthusiasts/docker-planefence/issues/211 
To do-- implementing this also for Plane-Alert (which has a bit more complex and potentially long CSV file structure, so i don't necessarily want to regenerate the entire RSS file everytime)

- RSS for planefence should be available at planefence.rss and planefence-yymmdd.rss
- file retention is managed centrally (default 2 weeks)
- URL and Favicon links can be managed with these parameters in `planefence.config`:
```bash
PF_RSS_SITELINK=""
PF_RSS_FAVICONLINK=""
PA_RSS_SITELINK=""
PA_RSS_FAVICONLINK=""
```