sdr-enthusiasts/docker-planefence #244: Adding Telegram Notification Support to Planefence / Plane-alert

smkrv, 2025-05-04:
First off, thank you for the outstanding Docker images and the care you put into this project - the whole stack runs smoothly right out of the box. After using PlaneFence for a while, I found myself wishing for a bit of extra functionality and thought it might be useful to share the idea with the community.

I'm submitting this as an issue rather than a PR since I'm not entirely familiar with the codebase architecture and how notification systems are integrated throughout the application. This is a proposal for adding Telegram notification support to docker-planefence, similar to existing notification methods like BlueSky and Mastodon.

## Overview

The implementation consists of:
1. A new `post2telegram.sh` script
2. Configuration parameters in `planefence.config`
3. Minor Dockerfile modifications to ensure dependencies

## Implementation

### 1. New Script: `rootfs/scripts/post2telegram.sh`

```bash
#!/command/with-contenv bash
#shellcheck shell=bash disable=SC1091,SC2174,SC2015,SC2154
# -----------------------------------------------------------------------------------
# Copyright 2025 Ramon F. Kolb - licensed under the terms and conditions
# of GPLv3. The terms and conditions of this license are included with the Github
# distribution of this package, and are also available here:
# https://github.com/sdr-enthusiasts/docker-planefence/
#
# This package may incorporate other software and license terms.
# -----------------------------------------------------------------------------------

source /usr/share/planefence/persist/planefence.config
source /scripts/common

if (( ${#@} < 1 )); then
    "${s6wrap[@]}" echo "Usage: $0 <text> [image1] [image2] ..."
    exit 1
fi

# Set the default values
TELEGRAM_API="${TELEGRAM_API:-https://api.telegram.org/bot}"
TELEGRAM_MAX_LENGTH="${TELEGRAM_MAX_LENGTH:-4096}"

# Check if the required variables are set
if [[ -z "$TELEGRAM_BOT_TOKEN" ]]; then
    "${s6wrap[@]}" echo "Fatal: the TELEGRAM_BOT_TOKEN environment variable must be set"
    exit 1
fi
if [[ -z "$TELEGRAM_CHAT_ID" ]]; then
    "${s6wrap[@]}" echo "Fatal: the TELEGRAM_CHAT_ID environment variable must be set"
    exit 1
fi

# Extract info from the command line arguments
args=("$@")
TEXT="${args[0]}"
IMAGES=("${args[1]}" "${args[2]}" "${args[3]}" "${args[4]}") # up to 4 images

if [[ -z "$TEXT" ]]; then
    "${s6wrap[@]}" echo "Fatal: a message text must be included in the request to $0"
    exit 1
fi

# Clean up the text
TEXT="${TEXT:0:$TELEGRAM_MAX_LENGTH}"      # limit to max characters
TEXT="${TEXT//[[:cntrl:]]/\\n}"            # Replace control characters with newlines

# Send images to Telegram if available
has_images=false
for image in "${IMAGES[@]}"; do
    # Skip if the image is not a file that exists
    if [[ -z "$image" ]] || [[ ! -f "$image" ]]; then
        continue
    fi
    
    has_images=true
    
    # Send the photo with the message
    response="$(curl -s -X POST "${TELEGRAM_API}${TELEGRAM_BOT_TOKEN}/sendPhoto" \
        -F "chat_id=${TELEGRAM_CHAT_ID}" \
        -F "photo=@${image}" \
        -F "caption=${TEXT}" \
        -F "parse_mode=HTML")"
    
    message_id="$(jq -r '.result.message_id' <<< "$response" 2>/dev/null)"
    
    if [[ -z "$message_id" ]] || [[ "$message_id" == "null" ]]; then
        "${s6wrap[@]}" echo "Error sending photo to Telegram: $response"
        { echo "{ \"title\": \"Telegram Photo Send Error\","
          echo "  \"response\": $response }"
        } >> /tmp/telegram.json
    else
        "${s6wrap[@]}" echo "Photo message sent successfully to Telegram with message_id: $message_id"
        echo "$message_id" > /tmp/telegram.link
        # Only send the first image with text, then exit
        exit 0
    fi
done

# If no images or image sending failed, send text only
if ! $has_images; then
    response="$(curl -s -X POST "${TELEGRAM_API}${TELEGRAM_BOT_TOKEN}/sendMessage" \
        -F "chat_id=${TELEGRAM_CHAT_ID}" \
        -F "text=${TEXT}" \
        -F "parse_mode=HTML")"
    
    message_id="$(jq -r '.result.message_id' <<< "$response" 2>/dev/null)"
    
    if [[ -z "$message_id" ]] || [[ "$message_id" == "null" ]]; then
        "${s6wrap[@]}" echo "Error sending message to Telegram: $response"
        { echo "{ \"title\": \"Telegram Message Send Error\","
          echo "  \"response\": $response }"
        } >> /tmp/telegram.json
        exit 1
    else
        "${s6wrap[@]}" echo "Text message sent successfully to Telegram with message_id: $message_id"
        echo "$message_id" > /tmp/telegram.link
    fi
fi
```

### 2. Configuration Parameters for `planefence.config`

Add the following section to the configuration file:

```bash
# ---------------------------------------------------------------------
# These are optional parameters related to Telegram notifications
# To enable Telegram notifications, create a bot using BotFather and obtain a bot token
# Then get your chat ID (you can use @userinfobot or other methods)
# TELEGRAM_BOT_TOKEN should look like "123456789:ABCDefGhIJKlmNoPQRsTUVwxyZ"
# TELEGRAM_CHAT_ID can be a numeric ID or a channel name like "@yourchannel"
#
# The parameters "PF_TELEGRAM_ENABLED" and "PA_TELEGRAM_ENABLED" must be set to "on"/"enabled"/"1"/"yes" to start
# notification about Planefence and Plane-Alert respectively.
TELEGRAM_BOT_TOKEN=""
TELEGRAM_CHAT_ID=""
PF_TELEGRAM_ENABLED=""
PA_TELEGRAM_ENABLED=""
```

### 3. Dockerfile Modifications

The script requires `jq` for JSON processing. If `jq` is not already included in the base image, the following changes to the Dockerfile would be needed:

```dockerfile
# In the KEPT_PACKAGES array, add jq:
KEPT_PACKAGES+=(jq) && \
```

And in the `chmod` command section, add the Telegram script:

```dockerfile
chmod a+x /usr/share/planefence/*.sh /usr/share/planefence/*.py /usr/share/planefence/*.pl /scripts/post2telegram.sh && \
```

## Features

The implementation:
- Supports sending both text and image messages to Telegram
- Handles multiple image files (sends the first valid image)
- Provides error handling and logging
- Implements HTML parsing mode for rich text formatting
- Stores message ID for potential future reference
- Follows the same pattern as existing notification methods

## Integration

I'm not sure about the exact integration points with the main planefence and plane-alert scripts. This would need to be integrated to trigger when aircraft are detected, similar to how other notification services are triggered. I've modeled this script after the existing notification scripts like `post2bsky.sh`, but I'm not familiar with how these are called from the main application flow.

## Dependencies

- `curl`: For API communication
- `jq`: For JSON response processing

I hope this implementation is helpful! If there are any specific architectural considerations or integration points I've missed, please let me know and I'd be happy to refine the approach.

kx1t, 2025-06-18:
Hi, thanks for your suggestion and sample implementation code. Sorry for the late reaction.

I’ll take a look later this week/next week to see if and how to add this.

kx1t, 2025-06-20:
Made a bit of progress on this -- it's ready for beta-testing in a branch. (Note - in the future, once we merge the functionality to `main`, we'll remove the branch and you'd have to switch back to the `latest` tag)
1. change the Planefence image: line to this build statement:
```yaml
    build: https://github.com/sdr-enthusiasts/docker-planefence.git#telegram
```
2. Follow the instructions to set things up from here: https://github.com/sdr-enthusiasts/docker-planefence/blob/telegram/README-telegram.md
3. If you want to see some live action from my testing setup, take a look at my testing channel: https://t.me/+B_r-DgeNEQ4yMTUx

[image]

closed 2025-06-22 by kx1t